makermoekoe/Picoclick-C3 #3: Not espressif but impressive

pshobowale, 2022-01-11:
1.) On the question if you could improve the performance:
Are there still boot messages if you look at the serialport? Because if there are you could try to completely disable them as this wastes time during startup. --> https://esp32.com/viewtopic.php?t=24244 and https://www.esp32.com/viewtopic.php?t=1658 (don't know if it is the same procedure as for the c3)

2.) How much longer is the boot time if you validate the boot image, because this could be a possible backdoor?

3.) May I use you your design for a private project? (I want to try to build a kind of opensource alternative to this --> https://freedrum.rocks/)

septatrix, 2022-01-19:
You could also take a look inside the [`esp_system/startup.c`](https://github.com/espressif/esp-idf/blob/1d02e638bd05b80d6fd648d51c14a8ecd6c2dbdc/components/esp_system/startup.c)¹ file and try some of the other hooks there. It seems like using `__attribute__(constructor)`, `__attribute__ ((section(".initX"))` or `ESP_SYSTEM_INIT_FN` should all provide you with earlier places to enable the power latch. The earliest hook seems to be to define a custom `start_cpu0` and inside that call `start_cpu0_default` to continue the normal boot process. I am not quite sure if some initialization has to be performed before the gpio is usable via the usual functions so some of the more drastic/early hooks might not work or you would need to write to the raw registers instead. Anyway if you want to register even shorter button presses this might help. If you have any question feel free to ask me (also in german ;D).

¹ Potentially also [`esp_system/port/cpu_start.c`](https://github.com/espressif/esp-idf/blob/1d02e638bd05b80d6fd648d51c14a8ecd6c2dbdc/components/esp_system/port/cpu_start.c) though I doubt you can hook into there - also the system is very much uninitialized that early.

makermoekoe, 2022-01-24:
There aren't any messages while the ESP is booting anymore. Nevertheless, the easier way is to disable them completely, yes...

I don't now it anymore because I have tested so many things on that day back then. I should definitely measure it again and check if it's worth it.

And as I have not yet set any license here you can actually use whatever you want!

makermoekoe, 2022-01-24:
> You could also take a look inside the [`esp_system/startup.c`](https://github.com/espressif/esp-idf/blob/1d02e638bd05b80d6fd648d51c14a8ecd6c2dbdc/components/esp_system/startup.c)¹ file and try some [...]

You genius. Initially the voltage divider workaround and then the idea with the early switching latch haha! Yes, I have read something about the second stage bootloader and that it is possible to do actions there as well. Regarding the serial output while booting this action would take place after 27ms. And when disabling serial output this would even be more early. But as I'm new to the ESP-IDF this is a task in the future only... 

Not sure if you have ordered a Picoclick, but if yes it would be nice to get support in this 🙂

septatrix, 2022-01-24:
> You genius. Initially the voltage divider workaround and then the idea with the early switching latch haha!

Thanks, I feel flattered :)

> Not sure if you have ordered a Picoclick, but if yes it would be nice to get support in this 🙂

Not currently, no. I wanted to wait if there might be some hardware revision should the voltage divider be improvable or maybe even slightly modify the board to allow for two buttons and build it myself to save a few bucks.

However I have a normal ESP32 lying around with which I could try some stuff. I won't be able to get any absolute timings as I have no access to an osci but I already have some other ideas in mind to get relative timings.

septatrix, 2022-01-24:
> And as I have not yet set any license here you can actually use whatever you want!

As far as I know you retain every right to the content in this repo without a license meaning that redistributing, modifying etc is prohibited (https://opensource.stackexchange.com/a/1721). I am not well versed with hardware licenses though there should also be some permissive ones you might consider choosing from.

septatrix, 2022-02-12:
The entry points I initially planned to target only yielded small improvements of a few milliseconds at most while being very hacky and using undocumented/internal apis.
However the second stage bootloader seems very promising as it exposes a before/after hook which can be used. The `bootloader_before_init` hook fires after only 850210 cycles on my standard ESP32 which would be about ~5ms@160MHz. The RISC-V core will differ slightly though likely not by much. Sadly I have not been able to convince PlatformIO to also build the second stage bootloader but when directly using idf.py everything seems to work flawless.
Next steps would be to configure some IO pins as the normal GPIO component is not yet initialized so we have to access the registers directly. This should be fairly simply.

I also made an issue about more init hooks before stumbling upon the bootloader hooks where @igrr also suggested simply using a GPIO pin which defaults to a weak pull-up. This could then be made into a low output to disable the latch again. I do not quite grasp the whole latching mechanism and therefore do not know if such a weak pull-up would be sufficient (or if a pull-down would be required which does not seem to be the default state for any of the pins). For anyone interested here the issue: https://github.com/espressif/esp-idf/issues/8390

closed 2022-04-19 by makermoekoe