makermoekoe/Picoclick-C3 #8: Is the battery voltage checked anywhere?

septatrix, 2022-01-20:
In the README it says that most standby current is used by the voltage divider connected to the battery which according to the schematics is connected only to the ESP32. However in the code I can find no usage of that pin except for the defines. Is there maybe some other place where this information is used? I plan to create my own iot button inspired by you project and if it is not need I consider dropping the voltage divider.

makermoekoe, 2022-01-21:
Well, the battery divider is not needed when don't care about the battery state. It's there to read the batteries voltage level und thus estimate the state of the battery (like in percentage). You could, for example, disable WiFi if the battery state is too low to protect it from being overdischarged, or it least discharged too much, because there is a battery protection which protects it from overdischarge. Another example is to send the battery state out so that you can visualize it somewhere else, like it is also done with smart home remotes or PIRs.

In the included sketches the battery state is not used yet, yes, but I will update this as soon as possible (hopefully on this weekend). It's as easy as reading the GPIO4 (ADC1_CH4) and convert it to a voltage with consideration to the divider.

septatrix, 2022-01-21:
Thanks for the reply. So if it is only used by the ESP32 would it not be possible to put a low-leakage transistor between the voltage divider and ground or connect the ground of the voltage divider to a pin of the ESP32 which could be pulled low when needed¹? Given that it draws a majority of the standby current it would increase the lifetime per charge significantly.

¹ This would of course only work if the GPIO would be in input or high-impedance mode when unpowered though I have not yet found official information regarding that behaviour.

makermoekoe, 2022-01-24:
For the first version of the Picoclick (not the C3 version) I tried to connect the voltage divider to a GPIO on the GND side and turn it low while reading the voltage. Somehow I took a wrong GPIO (probably a floating or a default low one), which was so frustrating that I didn't reconsider working on this again... 
But now, as it is clear that it was just a wrong chosen GPIO, I will definitely try to implement something like this again! 3µA are still too much... 

Btw. I have updated the code which can now read the battery voltage. It is as easy as this:

Setup:
`#define ADC_PIN ADC1_CHANNEL_4
  adc1_config_width(ADC_WIDTH_12Bit);
  adc1_config_channel_atten(ADC_PIN, ADC_ATTEN_0db);`

and to read the voltage (in mV):
`int voltage_mv = adc1_get_raw(ADC_PIN);`

septatrix, 2022-01-24:
> For the first version of the Picoclick (not the C3 version) I tried to connect the voltage divider to a GPIO on the GND side and turn it low while reading the voltage. Somehow I took a wrong GPIO [...]

Isn't a floating pin exactly what you want or more specifically high impedance (the third state apart from input/output)? Output-low would not yield an improvement, output-high would probably either damage the ADC pin when the μC has power or at least be equivalent to output-low without power. Input on the other hand might also damage the controller? Though I am tapping a bit in the dark here as one does not deal with powerless μCs every day and just imagining what might be possible. High impedance should behave as close to an unconnected pin as possible though I am also not sure what voltage the ADC would be exposed to in that scenario.

The safest bet would probably be a transistor with a big pull-down between the voltage divider and GND (as well as maybe also the midpoint and ADC - again no clue what voltage would be there otherwise).The base of the transistor(s) would be connected to an output-low, input or high impedance pin of the ESP32 with a big pull-down in the latter cases to prevent the pin from floating. As such the pd should not draw significant power. It might also be possible to use a single transistor at the high-voltage side of the divider though I am again out of my comfort zone here.

makermoekoe, 2022-01-25:
> Isn't a floating pin exactly what you want or more specifically high impedance (the third state apart from input/output)? Output-low would not yield an improvement, output-high would probably either [...]

Thank you so much for sharing your thoughts, this is a huge help!

I've just checked the [ESP32 C3 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf) and on page 10/11 it seems like every GPIO of that CPU is a high impedance one (marked with a T)?!

I should definitely give it a try on the breadboard. Either this and the transistor solution.

septatrix, 2022-01-25:
> Thank you so much for sharing your thoughts, this is a huge help!
> 
> I've just checked the [ESP32 C3 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf) and on page 10/11 it seems like every GPIO of that CPU is a high impedance one (marked with a T)?!

Yes they should all be capable of high-impedance though I am not sure which state is active during boot. Maybe the [ESP32-C3 Technical Reference Manual (Section 5.11)](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf#subsection.5.11) if of a bit more help. It talks about the state during reset though I am not sure if this also applies to the powered down state.

> I should definitely give it a try on the breadboard. Either this and the transistor solution.

I think the transistor solution is better. Otherwise if the low side and midpoint of the voltage divider are connected to high impedance pins the whole line should(?) be at the voltage level of the battery which might still damage the pins (https://electronics.stackexchange.com/a/491928).

septatrix, 2022-01-27:
Sadly I do not have the components on hand but maybe the following can help you:

* [Battery Voltage Monitoring (StackExchange)](https://electronics.stackexchange.com/a/424634/306001)
* https://jeelabs.org/2013/05/17/zero-powe-battery-measurement/
* https://jeelabs.org/2013/05/18/zero-power-measurement-part-2/index.html

makermoekoe, 2022-01-27:
Especially the idea behind the third link looks very interesting! Just a short period of time to take the measurement but that should be fine, I like it.

I will give it a try on the weekend and keep you updated!

makermoekoe, 2022-02-03:
Ouuh yes! It works. 
This problem finally made me buy an oscilloscope 😄 I received it yesterday and initially tested it with the third point of your list above. It does exactly what it should do. With a 1M pullup resistor to the pmos, I got 196ms where the ADC can be read. With a 100k I got 20ms and so on.

Next I will try to hook it up to the Picoclick completely (the mosfet control was done with a jumper only, but that shouldn't be a problem)

septatrix, 2022-02-04:
Didn't you have one in your video or did you borrow that one? Anyways glad to see that it worked as I have not been able to test it myself.

For everyone who is curious I made a simulation for the [circuit with a transistor](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWEAmaYAcBmMkFgCwYBsyk6+RIJIS+mNApgLRhgBQA5uAJzIj5weffPnRQobAEpDwRSDO6V58gTXHLoCNgCcZA+WF4hMpcTnhwdMkwaPJVBi5YDu108gDsfGxIAOKL2N3QP1wdTZXQ2FBT2jlNgAzFARKZHsZNPxjejBNKFhkCIyyGTA5CUijMvlkFNl43VrKauTU9McnIqaUdOR0MUyJABds1NU+gYcwpnxoDw9uclwPbCIibkh6GGRDNbKsSG4KE2bBABMGBIBDAFcAGyHOFH7W54HAhtGUVUIx5XALFJ+Ogaj8QeAPEpxKpMNAtupNGwAM78MrfFTg+z-CDXO5IhhsAD2ajRKkOAzyMAgBiJYTR9AE3A8ahgFiZRBStTCfCQkDYQA) as well as for the [circuit with a capacitor](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWEAmaYAcBmMkFgCwYBsyk6+RIR+IS+mtApgLRhgBQA5uAJzIj44vfvnzooUdgCVh4IpFk9KChYNoSV0BOwBOswQrB8QmUhJzw4u2acPHkaw5asB3G2eQB2frckAHFG8TDyCDcA12NyMRIS8YlXYAMxQESmQHWXSaTAYwLShYZEjMslkweUko43KFZFS5BL06yhqUtIynZ2LmlAzkdHEsyQBjFAHe7Mx2mkMWTxNoT3KMQVJyfDrPZVhIDhkwtYEhAhmJNUxoBhUC7W4DtQPxhIBlEymJo9qO8MSAQwAbADOjAkRQA9upymdIDxBvkYBBDOwIRAoQxBDx5kgYJZMURUnVwvxscjwPwoaoYTx1Dj4HiCUh+OSfOwgA)
The circuit with a capacitor needs some time to charge the capacitor before a measurement can be taken which the transistor circuit does not require. The transistor circuit also seems to be able to handle floating connections better. Apart from that they should function very similar. It might be interesting to compare the leakage current for both of them. Should you have the components lying around it might be interesting to compare with the Otii Arc ;D

[image]

[image]

septatrix, 2022-04-06:
Hey, just wanted to check in whether you worked on this the last two months? :)

makermoekoe, 2022-04-07:
How time flies 😰 
But yes I have worked on it for the last two months! Not sure if you're following on Instagram, but my latest board (the purple Paperclick one) has both the new battery monitoring feature with a single mosfet and the latching circuit with VDD_SPI on board. Second one is already tested and works like a charm! Have to do some scope measurements but so far I haven't got any unsuccessful button presses. Once calling deepsleep the device turns itself off. And the best: if you're not in deepsleep the device will keep its power while flashing, so the button doesn't need to be pressed during the upload.

The battery monitoring will be tested latest on the weekend. I have added a couple of test points to the board in order to make a few measurements. The only thing I can tell you right now is that we're at 120nA "sleep" current (measured with the Power Profiler Kit 2). 

In parallel I am working on a C3T v2 :)

septatrix, 2022-04-07:
That's great to hear. Can't wait for a new video :)

Nauman3S, 2022-10-16:
How long the battery could last on a single charge? Maybe one month?

haldi4803, 2022-10-30:
> 

Going by the 11 MicroAmps in Deepsleep (with Motion Sensor) and the 30mAh Battery used that would be around 113 Days Deep Sleep only.