makermoekoe/Picoclick-C3 #10: port_common.c:129: undefined reference to `app_main'

Bibbbi, 2022-01-26:
I use ESPRESSIF IDF v4.3.2 for software development. This allows the **Picoclick_C3T_hwtest_v2** example to be compiled successfully.

With **ESPNOW_master** and **Picoclick_ESPNOW_slave** I always get the following error.

`FAILED: ESPNOW_master.elf
cmd.exe /C "cd . && C:\Users\Markus\Documents\esp\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\ESPNOW_master.elf.rsp  -o ESPNOW_master.elf  && cd ."
c:/users/markus/documents/esp/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x14): undefined reference to `app_main'
c:/users/markus/documents/esp/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
C:/Users/Markus/Documents/esp/.espressif/frameworks/esp-idf-v4.3.2/components/freertos/port/port_common.c:129: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

makermoekoe, 2022-01-28:
For the ESP32-C3 you will need to update your esp-idf to v4.4-rc1. Would be nice if you can try this and let us know if this will help.

I've uploaded both examples (espnow slave & master) for building without PlatformIO. May you wanna try this as well.

Bibbbi, 2022-01-29:
The newer [build](https://github.com/makermoekoe/Picoclick-C3/tree/0d348d197d335b6a68c77b21edbe15e7a65e43cf) has fixed the problem. You can close this issue. Thank you.

closed 2022-01-29 by Bibbbi