makermoekoe/Picoclick-C3 #11: Pick and Place for PCBway?

Fedack, 2022-02-11:
Greetings!
Considering your store is often out of stock of those, I wanted to know if you'd be willing to add the pick and place for PCBway as one of the files?
I'm sadly not good enough yet in soldering to work with such small components but I could actually use such a small IOT Button.

Thanks,
Fedack

Fedack, 2022-02-11:
I'm stupid, I just noticed they are in the assembly file. Ignore this

closed 2022-02-11 by Fedack

makermoekoe, 2022-02-11:
Great you found it. Just to let you know, I've worked on a restock of Picoclicks the last days, which will probably take place on this weekend. Unfortunately I'm not able to provide the amount of boards which are required, meaning they will be sold out again very fast.

Fedack, 2022-02-11:
Well, I joined the wait list for a couple. I saw the cost on pcbway and
since it's such a short run I don't think it's worth spending 200$ cad+ for
5 of the things.