makermoekoe/Picoclick-C3 #14: guidance with software

Zixim, 2022-02-17:
So I just received a batch of 3 Picoclick-C3 devices.
The device is really well made, great job @makermoekoe !!

The thing I'm now stuck with is that I don't see an easy path for integrating the devices into my Home Assistant setup?

- I have many ESPHome-based devices around the house, but that doesn't look like a great option, ESPHome will likely kill the battery life - even with deepsleep.
- MQTT might be an option, but I have zero experience with the Arduino-IDE...
- ... ?

So, is anyone using the Picoclick-C3 with Home Assistant? How?

Failing anyone with Home Assistant experience, what would be the best way to achieve this ?

makermoekoe, 2022-02-17:
Great to hear so, thank you!

Unfortunately I've either not worked with Home Assistant nor with ESPHome yet, but we will somehow get it working. MQTT is probably one of the easiest ways to get it to work, but the Picoclick would get a delay from the button press to sending its message because it has to connect to the router first. 

How do you program your ESPHome devices? Is it possible to equip one of these devices with a modified/custom code? Then the best solution would be to act one of those as an ESPNOW master device, which will act as a broker between your Picoclicks and your smart home. In this case you will get an immediate response in your smart home like shown in the Youtube video of the Picoclick.

If this isn't possible you can set up an extra ESP device to be the broker/bridge between the Picoclicks and the smart home.

Zixim, 2022-02-17:
I've been using a few of these : [Hugo-ESP8266](https://github.com/mcer12/Hugo-ESP8266) , they are using MQTT, and I'm very happy with the latency. But they are using a batt with much higher capacity.
No longer available, so I went looking for alternatives & came across Picoclick.

ESPHome compiles a yaml file and offers to upload via OTA, or USB, or download the binary fw and use a flasher software on your computer.
People have been trying to include an ESPNOW component into ESPHome, but it isn't there (yet)... <https://github.com/iphong/esphome-espnow>

"...set up an extra ESP device to be the broker/bridge..." this would certainly be an option, have you got a specific tutorial in mind ?

Zixim, 2022-02-17:
Would you mind sending the bin files for flashing please ?

makermoekoe, 2022-05-25:
I've finally set up my first Home Assistant instance and I'm very pleased by it. So happy about it that I shifted all my smart home devices to work with HA. 
In addition to this I started to work with ESPHome as well, which seems to be a very useful tool for non-battery powered devices (haven't played with deep sleep in ESPHome at all). Thus I think integrating the Picoclicks to ESPHome is not the best solution.

Nevertheless I wanted to integrate them to HA so I tried this by using the MQTT plugin. But as I want to use ESPNow for the Picoclicks because it is way faster and extends battery life I set up a ESPNow to MQTT bridge using an ESP32-S2 Thing board from Sparkfun (should be possible with any ESP based board). I published both the Picoclick code and the bridge code a couple days ago. 

After installing the MQTT plugin in HA I added this code to the `configuration.yaml ` which will create three entities for the Picoclick device (battery voltage, battery percentage and binary_sensor):

`sensor:
  - platform: mqtt
    name: Battery_Picoclick1
    state_topic: "home/node1234/battery"
    device_class: voltage
    unit_of_measurement: "V"
    value_template: "{{ value | round(0)}}"
  - platform: mqtt
    name: BatteryLevel_Picoclick1
    state_topic: "home/node1234/batterylevel"
    device_class: battery
    unit_of_measurement: "%"
    value_template: "{{ value | round(0)}}"

binary_sensor:
  - platform: mqtt
    name: Picoclick1
    state_topic: "home/node1234/value"
    payload_on: true
    payload_off: false
    value_template: "{{ value | int > 0 }}"`

All in all this is working brilliant. It is running a couple days now without any problems or interruptions. You won't even notice a delay between pressing the Picoclick and the actuation in HA. 

Btw. you don't even have to create entities for the Picoclick, you can use it within the automations of HA directly (via MQTT) which will looks like this for toggling an outlet:

`alias: Picoclick1_automation
description: ''
trigger:
  - platform: mqtt
    topic: home/node1234/value
    payload: '1.00'
condition: []
action:
  - type: toggle
    device_id: YOUR_DEVICE_ID
    entity_id: YOUR_ENTITY_ID
    domain: switch
mode: single`

Hope this helps. If you have questions then please let me know! I'm now on the road to become a HA pro 😄

Zixim, 2022-07-12:
I'd like to know exactly what firmware you are flashing ?