makermoekoe/Picoclick-C3 #19: how to flash this with ESPHome ?

Zixim, 2022-07-12:
So...I bought 3 pcs a couple of months ago...and I managed to brick 2 of them trying to flash ESPHome onto them.

After flashing serial log (Termite) shows : 
```
Using 'COM11' as serial port.
Connecting...
Detecting chip type... ESP32-C3
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-C3 (revision 3)
 - Number of Cores: 1
 - Max CPU Frequenc... (96 %)Writing at 0x000b8e68... (100 %)Wrote 709504 bytes (420762 compressed) at 0x00010000 in 9.8 seconds (effective 578.8 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[10:33:43]invalid header: 0xffffffff
[10:33:43]invalid header: 0xffffffff
[10:33:43]invalid header: 0xffffffff
[10:33:43]invalid header: 0xffffffff
...line repeats hundreds of times...
[10:33:44]invalid header: 0xffffffff
[10:33:44]invalid header: 0xffffffff
[10:33:44]invalid header:ESP-ROM:esp32c3-api1-20210207
[10:33:44]Build:Feb  7 2021
[10:33:44]rst:0x7 (TG0WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
[10:33:44]Saved PC:0x4004c0dc
[10:33:44]invalid header: 0xffffffff
[10:33:44]invalid header: 0xffffffff
[10:33:44]invalid header: 0xffffffff
...line repeats hundreds of times...
```

What is going on ?
Perhaps someone could post a detailed step-by-step ?

makermoekoe, 2022-07-12:
What do mean with brick to of the Picoclicks? How do they behave now?

Regarding ESPHome, may the configuration file for the device could help here. How do you want to integrate them to ESPHome? Via MQTT? I'm still not an ESPHome professional...

TechOverflow, 2022-07-12:
I had the same problem with a few new C3's, albeit programmed via Arduino IDE.
I believe the problem was that I was accidentally pulling up GPIO9 (boot).

Zixim, 2022-07-12:
brick as in impossible to flash anything into them. Well...1 of them keeps rebooting so the comm port disappears in Windows.
If I get the timing right, it does accept a flash.

This is the basic ESPHome conf that I'm trying to run on the PicoClick C3:
```yaml
esphome:
  name: c3

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "C3 Fallback Hotspot"
    password: "eHc3ejhcDMMy"
```
It flashes OK, but just does this : 
```
Using 'COM11' as serial port.
Connecting...
Detecting chip type... ESP32-C3
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-C3 (revision 3)
 - Number of Cores: 1
 - Max CPU Frequenc... (96 %)Writing at 0x000b8e68... (100 %)Wrote 709504 bytes (420762 compressed) at 0x00010000 in 9.8 seconds (effective 578.8 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[10:33:43]invalid header: 0xffffffff
400 repeats....
[10:33:43]invalid header: 0xffffffff
[10:33:44]invalid header:ESP-ROM:esp32c3-api1-20210207
[10:33:44]Build:Feb  7 2021
[10:33:44]rst:0x7 (TG0WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
[10:33:44]Saved PC:0x4004c0dc
[10:33:44]invalid header: 0xffffffff
...repeats...
```

makermoekoe, 2022-07-12:
> I had the same problem with a few new C3's, albeit programmed via Arduino IDE.
> I believe the problem was that I was accidentally pulling up GPIO9 (boot).

That is indeed a problem. What was the reason of pulling it up? I'm asking because it isn't connected to anything...

If you have experience in soldering you can fix it by soldering a jumper cable to pin 15 of the ESP32C3 chip itself. Maybe you can try it without soldering as well, but you have to pull GPIO9 low while the Picoclick boots up, which can be done with pressing a GND connection cable to pin 15 while pressing the button.

[image]

makermoekoe, 2022-07-12:
> This is the basic ESPHome conf that I'm trying to run on the PicoClick C3:

Then you probably don't toggle the latch and the Picoclick looses power with releasing the button... What about keeping the button pressed? What does happen?

Zixim, 2022-07-12:
Please does anyone have a working basic ESPHome binary, that includes `ap` (without password) and `captive-portal`?
This would allow me to flash any other ESPHome fw into the device.

Zixim, 2022-07-12:
Pulling pin15 (GPIO9) to GND allowed ESPHome to flash the firmware from dashboard.

Boot time is way too long so LDO fails to latch.
I've been trying to set CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON , but don't think that ESPHome has a mechanism for this.

e-noyau, 2022-10-07:
Came here because I just soldered a LiPo to a picoClick C3T I just pulled out of its sealed package and I was going to flash it with ESPHome.  But I don't see any sign of life from the device when I plug it in (No lights, no com port), this is with the button firmly pressed of course (With a clothespin!).

Should I wait for the battery to charge? Anything should be visible while the battery is charging?

Is flashing with ESPHome a bad idea? @Zixim did you manage to get it to work?

Zixim, 2022-10-08:
> Is flashing with ESPHome a bad idea? @Zixim did you manage to get it to work?

bad idea unfortunately.
I did get it to work, but the latency with esphome is wayyyyyy too high when pressing the button.

scottcopus, 2023-01-25:
@Zixim ESPHome _does_ have some deep sleep capabilities of its own. But I've never experimented with it yet to find out its latency. But I'm guessing it'll be longer compared to @makermoekoe's sketch source that's here in the Picoclick's repo.

BTW, @makermoekoe I'd love to see some USA sources for the Picoclick and maybe other devices, and options that include a battery & case... and maybe qty discounts, lol.  Keep up the great work!

Zixim, 2023-01-26:
@scottcopus I have extensively tested ESPHome with deepsleep, and have many battery-powered esp32 devices around the house. They are just about all sensors gathering data (temp, hum, etc...). Latency isn't an issue in that scenario.

When the sleeping device is a controller (buttons to control lights for instance) then anything above 1 or 2 seconds of latency is huge.

ESPHome simply doesn't give us access to the functions in the framework that allow us to reduce startup time.

scottcopus, 2023-01-26:
@Zixim Good to know you've done deepsleep and startup latency testing with ESPHome. I'm sad it's not already optimized. Maybe they'll eventually have those similar menuconfig compile-time optimizations that makermoekoe showed us in one of his YT videos to speed up the boot time. If there's not already an issue within that project to help with this, maybe one of you guys could post it? I wouldn't exactly know how to phrase it.

makermoekoe, 2023-01-27:
> @scottcopus I have extensively tested ESPHome with deepsleep

But not especially for the Picoclick I think? Using deepsleep on the Picoclicks is not possible. At least it cannot wake up from deepsleep, because once the devices enter deepsleep the latch will be pulled low, which deactivates the complete Picoclick (both for the C3 and C3T). A button press is necessary to activate them again.

> @Zixim Good to know you've done deepsleep and startup latency testing with ESPHome. I'm sad it's not already optimized. Maybe they'll eventually have those similar menuconfig compile-time optimizations that makermoekoe showed us in one of his YT videos to speed up the boot time. 

The latency of 1 or 2 seconds is not related to the boot-up time, but the WiFi connection process. In my videos I'm using ESP-NOW to communicate with other devices, or moreover an espnow-2-mqtt bridge which will forward the messages from the Picoclicks as MQTT messages (which can be handled from HA). With using espnow the wifi connection is not needed and thus the communication only takes under 200ms (from Picoclick button press to receiving the message on the other side).
If you haven't done yet, watch the last 2-3 mins of my new Youtube video where I show how to do it with the bridge.

Zixim, 2023-01-27:
"But not especially for the Picoclick I think?"
I did try to get ESPHome + deepsleep working on the Picoclick, but - as you say - this is impossible due to latency.

When I say latency, I mean the time that elapses from click to boot to wifi connect to mqtt message transmit. The latch gets pulled low before wifi connects. (Which is a great thing for the battery ofcourse)
ESPHome just isn't the right firmware for the Picoclick.

I did have a look at ESP-NOW, but am reluctant to spin up yet another system in the household, and the setup doesn't look very simple to me.

makermoekoe, 2023-01-27:
> When I say latency, I mean the time that elapses from click to boot to wifi connect to mqtt message transmit. The latch gets pulled low before wifi connects. 

But if you pressed the button for around 200ms the Picoclick stays on and then the main latency is due to the wifi connect. The boot-up time is the part between pressing the button and pulling the latch high. After the latch is high the power will stay on.

makermoekoe, 2023-01-27:
> I did have a look at ESP-NOW, but am reluctant to spin up yet another system in the household, and the setup doesn't look very simple to me.

Up to you... It is simply uploading the ESPNOW2MQTT bridge code from the [Picoclick Gitbook](https://makermoekoe.gitbook.io/picoclick-c3/software/esp-now) (with your credentials) to any ESP, take the sender sketch for the Picoclicks and add a new MQTT device in HA. For me it couldn't be simpler at all.

Zixim, 2023-01-27:
here's me adding this to the give-it-a-go list.

closed 2023-01-27 by Zixim