makermoekoe/Picoclick-C3 #26: Value for capacitor C5 on the Picoclick c3

piningforfjords, 2023-02-27:
I noticed there is no value for the capacitor C5.  Was this something you were tweaking for the antenna?  Did you find a good value?

makermoekoe, 2023-02-28:
Actually 0Ohm works great. But haven't measured for any better solutions...

closed 2023-02-28 by makermoekoe

piningforfjords, 2023-03-01:
Ahh thanks for getting back to me.  I noticed you saying that in a comment  somewhere.  I didn't put it together because his is marked as a cap.  I won't make another issue that C5 should be marked as a resistor but someone might later.  I greatly admire your work.

chrisnet123, 2023-09-06:
Looking at picoclick_c3_v1.1.pdf C5 has no value. Is this a capacitor or a resistor? What is its value? Thanks! I love the board design and can not wait to adjust it!

makermoekoe, 2023-09-07:
Sorry, C5 is a part of the PI-matching network of the antenna. But for this board C5 should not be mounted. The other two components of the network are enough.

@piningforfjords I thought you are talking about the C3T where the antenna needs a 0ohm resistor in series, but for the C3 the parallel capacitor C5 should not be placed.

chrisnet123, 2023-09-07:
Thanks for the clarification. For the Picoclick C3  "C5" does not get placed! so it only needs C17 and C18. If I look at https://github.com/makermoekoe/Picoclick-C3/blob/main/docs/pc3_production.jpg and zoom in, I see it is not placed..... Thanks.