makermoekoe/Picoclick-C3 #27: Question about ESP32-C3 built-in USB

windorey, 2023-03-04:
Hello, sorry to bother but I am making my own ESP32-C3 board and now I am concerned about one thing.
Can I flash the board via USB by just connecting my USB connector to GPIO 18 and 19? Do I need to hold some pin high or low to upload specifically? Thanks in advance for any reply.

TechOverflow, 2023-03-04:
You don't normally need boot or reset to flash the binary via USB.
The issue I find is that for some reason, the C3 chips show a header error from the factory. This means that for the very first flash, you need to hold boot low before powering up (before plugging in USB) to boot into "upload mode". So you definitely want to have the option to short IO9 (boot) and EN (reset) to ground. On most ESP boards, this is done with buttons, but personally, since boot/reset are virtually never needed (especially if you intend on flashing OTA) I just place pads on the board so that I can short them with tweezers or something, instead of having to place actual buttons on the board.

windorey, 2023-03-04:
Does this also apply for ESP32-S3?

TechOverflow, 2023-03-04:
I have not had any projects with S3 modules, only S3 dev board, but I can imagine it's similar. Just make sure to check the datasheet for which IOs are strapping pins. For example on ESP32-C3 boot is IO9 to gnd, on ESP32 (original) boot is IO0, on ESP32-S3 boot is also IO0. Reset is always CHIP_PU (aka EN).
Best to check the schematics for one of the dev boards.

windorey, 2023-03-04:
Thank you. Also, for the ESP32 series, what are the XTAL pins, what I need to connect to them, and how do i find the value of what I need to connect to them?

NilsBe, 2023-03-05:
See page 15 of the esp32-c3 datasheet:

https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf

[image]

NilsBe, 2023-03-05:
> See page 15 of the esp32-c3 datasheet:
> 
> https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf
> 
> [image]

> Hello, sorry to bother but I am making my own ESP32-C3 board and now I am concerned about one thing. Can I flash the board via USB by just connecting my USB connector to GPIO 18 and 19? Do I need to hold some pin high or low to upload specifically? Thanks in advance for any reply.

See page 15 of the esp32-c3 datasheet:

https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf

[image]

NilsBe, 2023-03-05:
> Hello, sorry to bother but I am making my own ESP32-C3 board and now I am concerned about one thing. Can I flash the board via USB by just connecting my USB connector to GPIO 18 and 19? Do I need to hold some pin high or low to upload specifically? Thanks in advance for any reply.

See page 15 of the esp32-c3 datasheet:

https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf

[image]

makermoekoe, 2023-03-06:
> Can I flash the board via USB by just connecting my USB connector to GPIO 18 and 19? Do I need to hold some pin high or low to upload specifically? Thanks in advance for any reply.

Short answer: Yes, it is as easy as connecting those pins directly to the use connector. For me, even the boot strapping pin (for the C3 it is GPIO9) doesn't held to be low while uploading for the first time. If there's an header error incoming, you can solve it but just trying a second upload, which worked in almost every situation for me.
As @NilsBe mentioned, GPIO2 and GPIO8 should always be pulled up externally.

> Does this also apply for ESP32-S3?

For my S3 projects, I always put Boot and Reset buttons on the board. When entering the download mode another USB port will be opened (it is a real USB CDC port on the MCU side). Automatic download entering is (compared to the C3 and as far as I now) not possible for the S3 (and S2 either).

windorey, 2023-03-06:
I noticed the ESP chips need an external crystal and antenna, how to pick the right ones? I'm so sorry for asking so many questions.

makermoekoe, 2023-03-06:
Yes, the bare chips need both of them. But if you're not familiar with that it maybe is best to go with a module where everything is included. Modules with the ESP32-C3 are the ESP-C3-Mini-1 or the ESP32-C3-WROOM-02 just for example.

Else you need both crystal and antenna as the datasheet says. I'm using a 40MHz 2016 crystal with 9pF load capacitors and any 2.4GHz antenna with matching network according to your PCB layout. Currently using the SLDA31 or the PROANT440 3D antenna.

TechOverflow, 2023-03-06:
> [image]

Interestingly though, Espressif do not pull GPIO2 high in any of their C3 Dev boards.

windorey, 2023-03-06:
I notice a board "ESP32-PICO-KIT" has some "3D Antenna", I want to use it on my board. How can I find the part and use it in my schematic?

makermoekoe, 2023-03-06:
Yes, that one is the Abracon Pro-Ob-440 which I have mentioned above. Just create it on your own according to the datasheet, these are just four pads of which only two are connected. But it seems like there is also an ecad model on [Mouser].(https://www.mouser.de/ProductDetail/ABRACON/PRO-OB-440?qs=TCDPyi3sCW1HMjuU5AeiSw%3D%3D)

closed 2023-03-06 by windorey