makermoekoe/Picoclick-C3 #32: Add second button to schematic, with same wakeup function

AshwinDeTaeye, 2023-10-23:
Hello, 

I would like to add a second button to the schematic.
If possible have the same wake-up functionality as the first button.
Do I need to duplicate the below schematic completely for a second button?

[image]

makermoekoe, 2023-10-30:
Hey, you don't have to copy the whole circuit part. What's necessary for a second button are these components: SW1, R10, C19, R14 and D3. D3 is a dual diode, but you will only need one of them. The anode of the diode will be connected to the EN_LDO line.

closed 2023-10-30 by makermoekoe

AshwinDeTaeye, 2023-11-07:
Hello, thanks. Just to be sure could you check the modified schematics?

[image]