makermoekoe/Picoclick-C3 #33: Picoclick C3 not recognized in Windows 11?

tregota, 2023-10-29:
Do I need a driver for the picoclick to show up in PlatformIO win Windows11?
I have tried a variety of cables, the picoclick lights up but nothing shows up in platformio or in the device manager.

makermoekoe, 2023-10-30:
It's probably only the charge status LED which lights up. Do you have a suitable power supply connected to the battery port? A Lipo battery or something between 3.6V and 4.2V is perfect. 
If you don't wanna use an external power supply, you have to short a soldering jumper on the device: https://makermoekoe.gitbook.io/picoclick-c3/hardware

tregota, 2023-10-30:
Ah, sorry, I was not aware of this. Seems I made assumptions when I should have been rtfm. :) 
I'll try again with a battery soldered to it. Thanks.

closed 2023-10-30 by tregota